SpaceTeddy/CC1101 #1: Example code

Vild, 2016-03-02:
Hi, I'm interested in using this code for a project.
I don't fully understand the hardware and code, so some example code would be helpful.

Thanks
- Dan

o0Zz, 2016-03-21:
+1, one sample code would be nice.

Here is what I wrote, but my code is locked in cc1100_sent_packet function.

BR
```
#include "cc1100_arduino.h"

CC1100 cc1101;
uint8_t Rx_fifo[10];
uint8_t Rx_addr, Tx_addr, Pktlen, Lqi;
int8_t Rssi_dbm;
uint8_t Tx_fifo[20];
uint8_t My_addr;
uint8_t Patable[8];

void setup()
{
  Serial.begin(38400);
  Serial.println("start");

  uint8_t theRet = cc1101.cc1100_init();
  Serial.print("initializing: ");
  Serial.println(theRet);

  theRet = cc1101.cc1100_wakeup();
  Serial.print("wakeup: ");
  Serial.println(theRet);
}

void loop()
{
  
  while(1)
  {
    Serial.print("send: ");
    uint8_t buf[]={0xFE, 0xED, 0xBA, 0xBE};
    uint8_t theRet = cc1101.cc1100_sent_packet(0x00000000, 0x00000001, buf, sizeof(buf));
    Serial.println(theRet);
    delay(1000);
  }
}
```

SpaceTeddy, 2016-03-22:
Did you flash the eeprom file to the arduino first?

o0Zz, 2016-03-22:
Not at all, however i can see some default setting for eeprom. I supposed default settings should works, isn't it ?

I'm interrested having this file. can you share it ?

Thanks.

SpaceTeddy, 2016-08-15:
examples and eeprom utility is available now

closed 2016-10-30 by SpaceTeddy