SpaceTeddy/CC1101 #2: Building the CC1100 example

RossinesP, 2016-11-16:
Hi !
I am trying to build the example for the Raspberry but I'm a complete C++ noob. I copied all .cpp and .h in the raspi example, and called g++ *.cpp but I get this error : 

> cc1100_raspi.cpp:12:20: fatal error: cc1100.h: No such file or directory
>  #include <cc1100.h>

I naively believed the cc1100_raspi.h file was the one I was looking for, so I renamed it, but the build still fails.
Could you provide instructions to build the library please ? 
Thank you very much.

SpaceTeddy, 2016-11-17:
Well, it depends how you set up your file structure.

If you want it as simple as possible, please try the following:

- install wiringPi if not already installed. Try 
sudo apt-get install wiringpi 

If this does not work, follow the instructions on the wiringpi homepage.

- copy RX_demo.cpp and cc1100_raspi.cpp in the same directory.

- rename cc1100_raspi.cpp to cc1100.cpp

- open RX_Demo.cpp and change <cc1100.cpp> To "cc1100.cpp"

- open cc1100.cpp and de-comment CC1100 cc1100;

- compile RX_demo with sudo g++ -lwiringpi RX_demo.cpp cc1100.cpp -o RX_demo

I didn't tried it, but it should work

SpaceTeddy, 2016-11-17:
open RX_Demo.cpp and change <cc1100.cpp> To "cc1100.cpp"

RossinesP, 2016-11-19:
> open RX_Demo.cpp and change To "cc1100.cpp"

Thanks! I'll try that

SpaceTeddy, 2016-11-26:
does it work?

closed 2016-12-11 by SpaceTeddy

RossinesP, 2017-03-09:
I am really sorry I did not answer your question months ago. It does work well, thank you very much.