SpaceTeddy/CC1101 #3: Unable to get example running on RPi

computerlyrik, 2017-02-03:
Hi!
I have some problems getting started with a newly arrived CS1100 Board.

Here is what i did:
- Cloned your repo
- moved _rpi-Files to cs1100.cpp and cs1100.h
- copied over the RX-Demo for RPi
- `g++ cc1100.cpp RX_Demo.cpp -lwiringPi -o rx`

When calling the rx-program like: ./rx -f 434
I get the following error: `no CC11xx found!`

Debugged some around the code - also played with lower SPI-Speed and longer delays between reset-procedure - nothing. The Product and Version num remains on 0x00.

Hardware connections:
VCC --> RPi Header Pin 1
GND --> RPi Header Pin 25
SI     --> RPi Header Pin 19
SO   --> RPi Header Pin  21
SCK --> RPi Header Pin 23
GDO2 -->  RPi Header Pin 24

computerlyrik, 2017-02-03:
Hardware Probvem - Nevermind - can be closed.
Need to learn correct counting GPIO pins http://pi4j.com/pins/model-3b-rev1.html

SpaceTeddy, 2017-02-03:
means, after rewiring, the CC110x was found?

closed 2017-02-04 by SpaceTeddy