SpaceTeddy/CC1101 #5: No EEPROM settings...

chris250, 2017-04-03:
hi
just tried to run the tx_demo and it says that there is no eeprom setting...
(i've run the eeprom wirte sketch before)
hope you can help me.. (arduino rookie).
Chris

SpaceTeddy, 2017-04-04:
Hi Chris,

it is unfortunatly not enough to start the eeprom write sketch. 
please do the following:
- compile the tx_demo or rx_demo sketch
- remember the path of your compiled output data (arduino hex file and eep file)
- use the python eeprom_create.py to generate the  eeprom array for the eeprom_write.ino
  This is needed because the compile can choose the EEPROM position by its one.
  usage: ./eeprom_create.py <input *.eep file>
  you get an output file with *.array extension'''
- copy the output array content into the eeprom_write.ino sketch
- compile the eeprom_write.ino sketch
- upload to your hardware
- open the Arduino Serial console, set the baudrate to 38400 and restart your arduino hardware
- type the character w to the input field and press the sent button
- wait till eeprom is written
- sent r to verify that eeprom is written.

hope that helps

chris250, 2017-04-04:
thaks for your quick answer.
the eeprom_create.py is not working on my computer. seems that the window closes right after i've opend it.

SpaceTeddy, 2017-04-06:
do you have python installed and did you start the skript from the command line?

chris250, 2017-04-07:
haven't used python before so it might take a while to get in to it.

closed 2017-04-17 by SpaceTeddy

merlininthewoods, 2017-05-19:
Sorry to open this again but I am having a similar problem:

I have compiled, converted eep to arrray and flashed then compiled and uploaded Rx_demo.ino but still get "no EEPROM settings..."

It seems that 314 bytes are written to eeprom and I have checked they are actually written but then the begin() routine uses EEPROM_ADDRESS_CC1100_MY_ADDR which is defined as 0x1F6 or 506th byte (higher than the amount written!) to check if the eeprom data is there.  This is where it fails.

Arduino IDE 1.6.9
Arduino Pro mini board

I did swap PinChangeInt for EnableInterrupt  library because PinChangeInt was giving me problems during compilation but I don't think this would cause this problem as it doesn't get that far.

Any ideas

Thanks

SpaceTeddy, 2017-05-19:
hi,
may you read out your eeprom content with the eeprom sketch and post it here, please?
thx

merlininthewoods, 2017-05-19:
I used the default config for testing. Output from eeprom_write.ino after pressing 'r':

uint8_t eeprom = {
0x0b,0x1b,0x6d,0x67,0x50,0x85,0xc9,0xc1,0x03,0x17,0x1d,0x26,0x50,0x86,0xcd,0xc0,
0x6c,0x1c,0x06,0x3a,0x51,0x85,0xc8,0xc0,0x17,0x1d,0x26,0x69,0x51,0x86,0xcc,0xc3,
0x06,0x2e,0x06,0x47,0x57,0x43,0xff,0x04,0x05,0x00,0x00,0x06,0x00,0x21,0x65,0x6a,
0x87,0x83,0x3b,0x22,0xf8,0x15,0x07,0x30,0x18,0x14,0x6c,0x07,0x00,0x92,0x87,0x6b,
0xfb,0x56,0x17,0xe9,0x2a,0x00,0x1f,0x41,0x00,0x59,0x7f,0x3f,0x81,0x35,0x09,0x07,
0x2e,0x80,0x07,0x57,0x43,0x3e,0x0e,0x45,0xff,0x00,0x0c,0x00,0x21,0x65,0x6a,0x0e,
0x3b,0x73,0xa0,0xf8,0x00,0x07,0x0c,0x18,0x1d,0x1c,0xc7,0x40,0xb2,0x02,0x26,0x09,
0xb6,0x17,0xea,0x0a,0x00,0x19,0x41,0x00,0x59,0x7f,0x3f,0x81,0x3f,0x0b,0x07,0x2e,
0x80,0x07,0x57,0x43,0x3e,0x0e,0x45,0xff,0x00,0x0b,0x00,0x21,0x65,0x6a,0x2d,0x3b,
0x73,0xa0,0xf8,0x00,0x07,0x0c,0x18,0x1d,0x1c,0xc7,0x00,0xb2,0x02,0x26,0x09,0xb6,
0x17,0xea,0x0a,0x00,0x11,0x41,0x00,0x59,0x7f,0x3f,0x81,0x3f,0x0b,0x07,0x2e,0x80,
0x07,0x57,0x43,0x3e,0x0e,0x45,0xff,0x00,0x08,0x00,0x21,0x65,0x6a,0x5b,0xf8,0x13,
0xa0,0xf8,0x47,0x07,0x0c,0x18,0x1d,0x1c,0xc7,0x00,0xb2,0x02,0x26,0x09,0xb6,0x17,
0xea,0x0a,0x00,0x11,0x41,0x00,0x59,0x7f,0x3f,0x81,0x3f,0x0b,0x07,0x2e,0x80,0x07,
0x57,0x43,0x3e,0x0e,0x45,0xff,0x00,0x06,0x00,0x21,0x65,0x6a,0xca,0x83,0x13,0xa0,
0xf8,0x34,0x07,0x0c,0x19,0x16,0x6c,0x43,0x40,0x91,0x02,0x26,0x09,0x56,0x17,0xa9,
0x0a,0x00,0x11,0x41,0x00,0x59,0x7f,0x3f,0x81,0x3f,0x0b,0x07,0x2e,0x80,0x07,0x57,
0x43,0x3e,0x0e,0x45,0xff,0x00,0x08,0x00,0x21,0x65,0x6a,0xf5,0x83,0x13,0xa0,0xf8,
0x15,0x07,0x0c,0x19,0x16,0x6c,0x03,0x40,0x91,0x02,0x26,0x09,0x56,0x17,0xa9,0x0a,
0x00,0x11,0x41,0x00,0x59,0x7f,0x3f,0x81,0x3f,0x0b}
read done...

These are the exact same bytes the python script produced and I pasted into eeprom_write.ino in order to flash it

Fix needed in eeprom_create.py

74c74
< print ("#define EEPROM_LEN = " + str(total_bytes))
---
> print ("#define EEPROM_LEN " + str(total_bytes))

Seems to work well on the Rpi but I need to get the arduino working to test send and receive - thanks for your work!

SpaceTeddy, 2017-05-19:
i have updated the eeprom_write.ino tool.
may you retest, please?

thx

merlininthewoods, 2017-05-20:
That works:
Tx on Arduino:
```
TX_FIFO:060303000844FA
tx_time: 3ms
```

Rx on Pi:
```
RX_FIFO:0x06 0x03 0x03 0x00 0x08 0x44 0xFA 0x32 0x87 
RSSI:-53 LQI:0x07 CRC:0x00 
TX_FIFO: 0x05 0x03 0x03 0x41 0x63 0x6B 
Ack_sent!
```
Great!

Had to figure out what freq and modes where what as I have a 433 CC1101 board.  Is this correct:

```
Modes:
1 = GFSK_1_2_kb
2 = GFSK_38_4_kb
3 = GFSK_100_kb
4 = MSK_250_kb
5 = MSK_500_kb
6 = OOK_4_8_kb

Bands:
1 = 315
2 = 433
3 = 868
4 = 915
```
If so it might be good to include in the readme.

Thanks again

SpaceTeddy, 2017-05-20:
good to here that it is working, but you has set the TX address and RX address both to 0x03. This is not optimal. It would be better to define an unique Address for TX and RX. 

> Your TX data: TX_FIFO:06 **03 03** 00 08 44 FA
> 06 = length byte
> 03 = RX address
> 03 = TX address
> 00 08 44 FA = payload bytes
> 
> Better to have: TX_FIFO:06 **01** 03 00 08 44 FA
> 06 = length byte
> **01** = RX address
> 03 = TX address
> 00 08 44 FA = payload bytes

You're correct regarding modes and frequency. Sry that I did not described it in the ReadMe file.

merlininthewoods, 2017-05-22:
Thanks.  I have got it up and running reporting battery voltage from the arduino (addr:0x10) at my solar array to my Rpi (addr:0x01) in the house now.

All the best

SpaceTeddy, 2017-05-22:
perfect! ;)