SpaceTeddy/CC1101 #7: Wiring?

Ken98045, 2017-04-14:
Could you clarify the wiring setup you are using?  I think I understand it, but since I'm just getting started with the 1101 it is somewhat unclear since there appear to be some options.  Just a note in the readme or in a source comment would be very helpful.  Maybe I just missed it, but....

Also, a transmit example for the pi to go with the receive  example would be helpful too.

Thanks!

SpaceTeddy, 2017-04-15:
Hi Ken,
I have updated the readme with a pinout for this demo. 
I take a look for a TX demo for the Raspi, but the main loop should be very similar to the Arduino TX demo.
regards,
Chris

SpaceTeddy, 2017-04-15:
I created also an TX_Demo for Raspi.
Unfortunately I can not test it.

closed 2017-04-25 by SpaceTeddy