SpaceTeddy/CC1101 #8: Raspberry TX Hangs

esdad697, 2017-04-16:
To compile, changed includes in cc1100_raspi.cpp, RX_Demo.cpp, and TX_Demo.cpp to cc1100_raspi.h. 

TX_Demo seems to crash and don't see message in SDR#.

pi@raspberrypi:~/dev/CC1101 $ sudo ./TX_Demo -a1 -c1 -f868 -m250
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 4
Frequency: 3
Channel: 1
My_Addr: 1
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE9 
 #:0x01

SpaceTeddy, 2017-04-17:
Hi,
please be so kind and update the demo files and the cc1100_raspi.cpp and cc1100_raspi.h file and compile again.
It works on my desk now with 2 Raspbarry Pi's
thx and please give feedback

chris

SpaceTeddy, 2017-04-25:
Hi esdad697,

did you recompine check the new demo?
thx

closed 2017-05-02 by SpaceTeddy

esdad697, 2017-05-28:
With a few renames, everything compiles.