SpaceTeddy/CC1101 #9: Missing file (or maybe renamed) ?

kenji21, 2017-05-09:
First thanks for this "first steps" to use the CC1100.

Using Arduino Rx_demo example (had cloned your repo into my libraries one, and moved examples in their own directories).
But I now have this error:

````
.../libraries/CC1101/cc1100_arduino.cpp:11:20: fatal error: cc1100.h: No such file or directory
 #include <cc1100.h>
````

Tried to rename include inside cc1100_arduino.cpp file:

````
-#include <cc1100.h>
+#include <cc1100_arduino.h>
````

But got some other errors:

````
/Users/kenji/Documents/Arduino/libraries/CC1101/cc1100_arduino.cpp:356:60: error: no 'uint8_t CC1100::set_debug_level(uint8_t)' member function declared in class 'CC1100'
 uint8_t CC1100::set_debug_level(uint8_t set_debug_level = 1)  //default ON
                                                            ^
/Users/kenji/Documents/Arduino/libraries/CC1101/cc1100_arduino.cpp:365:37: error: no 'uint8_t CC1100::get_debug_level()' member function declared in class 'CC1100'
 uint8_t CC1100::get_debug_level(void)
                                     ^
````

What am I missing ?

SpaceTeddy, 2017-05-10:
you're right, seems to be an old Version in the repository.
I'll fix this hopefully todday.

SpaceTeddy, 2017-05-10:
please try again.

kenji21, 2017-05-10:
Nice,

Now just one error:

````
Rx_demo.ino: In function 'void setup()':
Rx_demo:39: error: 'class CC1100' has no member named 'silde'
   cc1100.silde();                          //set to ILDE first
````

And had to move some files in differents folder to have the arduino examples shown in the "File -> examples -> CC1100" appearing (in the Arduino "IDE"):

````
mkdir examples/Rx_demo
mkdir examples/Tx_demo
mv examples/Arduino/Rx_demo.ino examples/Rx_demo/
mv examples/Arduino/Tx_demo.ino examples/Tx_demo/
mkdir raspberry
mv cc1100_raspi.* raspberry/
mkdir raspberry/examples
mv examples/raspi/* raspberry/examples/
````

Also moved raspberry files in their own directory (to not be compiled by the Arduino IDE).

Do you want me to do a PR ?

kenji21, 2017-05-10:
there also is a missing " at #include "cc1100_arduino.h in cc1100_arduino.cpp:11

SpaceTeddy, 2017-05-15:
it is OK now?

closed 2017-05-22 by SpaceTeddy