SpaceTeddy/CC1101 #11: Not Receiving Anything

esdad697, 2017-05-28:
Not receiving with the lib on Raspberry Pi. 
Since two RPis are one foot apart, only left antenna on receiver.
Code default for GDO2 is SCL. Is this right?

sudo ./RX_Demo -v -a3 -c1 -f434 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 3
Frequency: 2
Channel: 1
My_Addr: 1
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x01 
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0 
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2 
0x02 0x26 0x09 0xB6 0x04 0xEC 0x2B 0x18 0x11 0x4D 
0x00 0x59 0x7F 0x3C 0x81 0x3F 0x0B 
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x02 
AND KEEPS echo out stuff

sudo ./RX_Demo -v -a3 -c1 -f434 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 3
Frequency: 2
Channel: 1
My_Addr: 3
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x03 
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0 
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2 
0x02 0x26 0x09 0xB6 0x04 0xEC 0x2B 0x18 0x11 0x4E 
0x40 0x59 0x7F 0x3C 0x81 0x3F 0x0B 
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0 

Can see signal at 434Mhz with SDR#, so something is being transmitted. RPi being used to receive can also transmit a signal visible on 434Mhz with SDR#.

Is GDO2 only used for RX? Are there some checks that can be run to verify CC1101 is setup right?
How should this be debugged?

SpaceTeddy, 2017-05-28:
Hi,
did you follow the description in the Readme file?
[ReadMe](https://github.com/SpaceTeddy/CC1101/blob/master/README.md)

> Code default for GDO2 is SCL. Is this right?
What do you mean with SCL? GDO2 is the GPIO pin that signals the raspi, that a package is ready to read from the CC1101 chip. 

> sudo ./RX_Demo -v -a3 -c1 -f434 -m100
> CC1100 SW: Verbose option is set
> Raspberry CC1101 SPI Library test
> Init CC1100...
> Partnumber: 0x00
> Version : 0x14
> ...done!
> Mode: 3
> Frequency: 2
> Channel: 1 [...]

here I see something strange. You started the RX_Demo tool, but it behaves like TX_Demo. May you confirm that is RX_Demo tool?

Please do the following:
- download the libs again from github
- follow the instructions in the ReadMe file
- increase the distance of TX and RX to around 3 meters. Less then 1m can disturb the RX circuit ( too much power).
-  start the tool with 

```
TX:
sudo ./TX_Demo -v -a1 -r3 -i1000 -t5 -c1 -f434 -m100
RX:
sudo ./RX_Demo -v -a3 -c1 -f434 -m100

```

esdad697, 2017-07-09:
I'm not sure of the pin assignments.

cc1100_raspi.h contains:
#define SS_PIN   10
#define GDO2	  6

The Raspberry Pi 3 GPIO Header shows pin 10 as GPIO15, but that results in:
Init CC1100...
no CC11xx found!

Setting SS to CE0 (GPIO8) fixes this. 

For GDO2, does pin 6 mean GPIO6? With the SparkFun Pi Wedge, these pins are all consecutive: CE0, MOSI, MISO, SCK, and RXI (GPIO6).

Rebuilt original source as follows:
git checkout .
git pull
cp examples/raspi/*
sudo g++ -lwiringPi RX_Demo.cpp cc1100_raspi.cpp -o RX_Demo
sudo g++ -lwiringPi TX_Demo.cpp cc1100_raspi.cpp -o TX_Demo

Placed Pis about 5 yards apart, and executed:
sudo ./TX_Demo -v -a1 -r3 -i1000 -t5 -c1 -f434 -m100CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 3
Frequency: 2
Channel: 1
My_Addr: 1
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x01 
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0 
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2 
0x02 0x26 0x09 0xB6 0x04 0xEC 0x2B 0x18 0x11 0x4C 
0x40 0x59 0x7F 0x3C 0x81 0x3F 0x0B 
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x03 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x04 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x05 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x06 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x0A 0x6D 
 #:0x01 
. . .

sudo ./RX_Demo -v -a3 -c1 -f434 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 3
Frequency: 2
Channel: 1
My_Addr: 3
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x03 
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0 
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2 
0x02 0x26 0x09 0xB6 0x04 0xEC 0x2B 0x18 0x11 0x4C 
0x40 0x59 0x7F 0x3C 0x81 0x3F 0x0B 
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0 

I take it this means still not working. Any suggestions on how to debug?

SpaceTeddy, 2017-07-09:
Hi,
This is strange that you have to change SS pin name to 8 to get SPI running. WiringPi is definitely using CE0 on pin name 10. Please check here http://3.bp.blogspot.com/-gmSvHB8OAHo/VRQvYSy5BLI/AAAAAAAABBQ/GHkqcVJEp4Q/s1600/Raspberry%2BPi%2BGPIO%2BPins.png

Are you sure that wiringPi is installed correct and activated? That pin name 8 is CE0 for original BCM setting.

Package receive pin is GPIO6. Pin name for wiringPi is 6. For BCM is 25.
You may try to set GDO2 pin to 25, but I suggest, that you check proper install of wiringPi first.

Regards 
Christian

esdad697, 2017-07-10:
WiringPi looks like it is installed right. I checked with "gpio readall". It is not working as expected or I'm missing something. I'm also using the SparkFun Pi Wedge. To fix, I wrote a blink utility that blinks an LED resistor pair connected to a pin.  Blink shows GPIO6 is connected to pin 22, and the other SPI pins are set right.

After updating GDO2 to 22, and rebuilding, started tx first.

sudo ./TX_Demo -v -a1 -r3 -c1 -f434 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 3
Frequency: 2
Channel: 1
My_Addr: 1
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x01 
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0 
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2 
0x02 0x26 0x09 0xB6 0x04 0xEC 0x2B 0x18 0x11 0x4E 
0x40 0x59 0x7F 0x3C 0x81 0x3F 0x0B 
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x08 0xAD 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x08 0xAD 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x0D 0x72 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x0D 0x72 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x12 0x37 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x12 0x37 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x16 0xFB 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x16 0xFB 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x1B 0xC1 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x1B 0xC1 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x20 0x85 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x20 0x85 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x25 0x4A 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x25 0x4A 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x2A 0x0E 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x2A 0x0E 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x2E 0xD2 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x2E 0xD2 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x33 0x97 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x33 0x97 
. . .

sudo ./RX_Demo -v -a3 -c1 -f434 -m100CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 3
Frequency: 2
Channel: 1
My_Addr: 3
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x03 
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0 
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2 
0x02 0x26 0x09 0xB6 0x04 0xEC 0x2B 0x18 0x11 0x4C 
0x40 0x59 0x7F 0x3C 0x81 0x3F 0x0B 
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0 

I take it nothing is being received, because RX_DEMO does not log anything.

What would be next step in debugging?

SpaceTeddy, 2017-07-10:
May you post a picture of you setup?
Which cc1101 receiver do you use?
Do you have an SDR receiver available to check that the transmitter is working?
May you connect your transmitter and receiver without SparkFun Pi Wedge?

esdad697, 2017-07-29:
I removed the SparkFun Pi Wedge. 

Sample run:
pi@raspberrypi:~/dev/CC1101 $ ls /dev/spi*
/dev/spidev0.0  /dev/spidev0.1

pi@raspberrypi:~/dev/CC1101 $ sudo ./TX_Demo -v -a1 -r3 -i1000 -t5 -c1 -f434 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
no CC11xx found!

So negative progress. Pinout matches expected and checked with blink program, so wiringPi is working.

CC1101 is from Amazon: [https://www.amazon.com/gp/product/B00NPZ3YUA/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1](url)

Raspberry Pi 3 Model B.

Pic of setup is attached. Red power pin is top right, because ribbon cable flips pinout.
[image]

esdad697, 2017-07-30:
Since it can’t find the SPI, it is something with WiringPi or the pinouts. I could be doing something wrong consistently with the wiring. I’d like to make my setup as close to yours as possible.

Are you using an RPi 3? Can you upload an image to a file sharing service like Google Drive and share with me. Github’s file attachment size limit is 25MB.
What CC1101 and cable are you using?
Upload images of connections to cable and CC1101 so I can compare with pinout.

esdad697, 2017-07-30:
I just ordered some F-F jumpers to avoid any cables. I assume jumpers are tight enough.

Is pinout right: 
[image]

SpaceTeddy, 2017-08-01:
Hi,
Attached my setup for a Rpi3

[image]

SpaceTeddy, 2017-08-22:
hi esdad697,

any news?

esdad697, 2017-09-03:
No luck.

Two RPis wired as in your pic. Pics of my RC RPi and CC1101 are attached. Since TX works, only wire that would seem worth checking is white G02. It is orange in your setup. TX and RX RPis are about 20 feet apart. CC1101 on RX was replaced with new one with spring antenna. TX was not changed as seems to work.

Used RTL-SDR to confirm both units can TX properly. Screenshot of signal from TX is attached. I’m not knowledgeable of GFSK, but seemed plausible. Take a look.

TX: sudo ./TX_Demo -v -a1 -r3 -c1 -f434 -m100
RX: sudo ./RX_Demo -v -a3 -c1 -f434 -m100

TX transmits every second:
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x8D 0x3F 
transmitted tx_timestamp: 36163ms 

RX hangs:
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x03 
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0 
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2 
0x02 0x26 0x09 0xB6 0x04 0xEC 0x2B 0x1A 0x11 0x4C 
0x40 0x59 0x7F 0x3C 0x81 0x3F 0x0B 
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0 

I wasn’t sure what RX should log, but assumed it should output something.

Both have GPIO supporting interrupts:
pi@raspberrypi:~/dev/CC1101 $ sudo python
Python 2.7.9 (default, Sep 17 2016, 20:26:04) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO
>>> GPIO.VERSION
'0.6.3'

Not sure what else to try. Would you have extra pair of RPi with CC1101 all wired up for sale or could I mail you mine?
[image]
[image]
[image]

esdad697, 2017-10-09:
Current theories are the CC1101 is being burned out by 5v levels on pin CSN or the CC1101 is receiving insufficient power.

SpaceTeddy, 2017-10-09:
Ok, but Chip Select pin should work, because you can read register settings. If you has the possibility you should by an new set of HW. 2x Raspi and 2 new CC1101 modules. 
Do you have 2 Arduinos, to check if the cc1101 modules are OK?

esdad697, 2017-10-09:
Have already tried new RPis, cc11101s, and Arduinos. Inability to Init spi with arduinos led me to 5v theory. Ti forum said some pins require lower voltage. Next will try level converter, already ordered, and probably replace cc1101s. I don't think arduinos or RPis are blown.

esdad697, 2017-10-29:
Four old CC1101s were burned out. Also replaced defective breadboard.

With Arduino, new CC1101, level converters, and ElectroDragon library sends, but does not receive sent messages from about 15'. With RPi, new CC1101, level converters, and SpaceTeddy library, sends, but does not receive sent messages. 

Both Arduino and RPi connect to CC1101 repeatedly, reliably based on logs. When transmitting about 2' to a SDR# with either platform and spring antenna, only about 1/3 of messages are received. With a 345 MHz Honeywell Sensor (the ultimate goal), transmissions are captured with SDR# reliably.

Since Arduino and RPi have same result, seems like issue with new CC1101s: defective hardware, incorrect hardware setup, or sofware setup. Setting the gain higher in the panstamp library did not seem to help. (cc1101.setTxPowerAmp(PA_LowPower);) Or issue could be with spring antenna or distance. RPis have not been replaced.

Could replace CC1101s with new ones with SMA antennas, or try setting gain in software. I don't see how this could be a RPi issue, but perhaps the SPI interface is only partly working.

Thoughts?

AutoDoc1955, 2018-01-02:
FWIW, I have the exact issue as esdad697.  CC1101 transmits OK, but I only see about 33% show up on SDR#.  See nothing coming from known-good Weather station transmitter as I see its beacon showing up in SDR#.  Still hacking and will report back here if I figure it out.

flashman21, 2018-02-14:
Hello, thanks for hardworking on this.
I traying to make it work, but have no luck.
Double checked wiring, compile, traying RX, but still see no CC11xx found.
I have one cc1101 module transmitting packets driving with microchip (working - checked with smartRF studio).
On my Rpi3 - SPI enabled in raspi-config, iringpi installed.
Only one thing I changed was register parameters in cc1100_raspi.cpp (cc1100_GFSK_1_2_kb) to be the same as mine TX module.
But this should not affect init comunication.
As I see the header, there is CC1100. Is there any problem to make it work with CC1101? Can I do any debug procedure?
Thank for any suggestions
Luk

flashman21, 2018-02-17:
it's working now, :-) Bad connect of CS pin.

xluccianox, 2018-09-06:
Hi, I think I have the same problem as @esdad697.
I'm using a Raspberry Pi Zero W and a TI-CC1101, connected like this:

[image]

When I run TX_Demo I can't see anything coherent in GQRX, at random periods I see some high gain packets beeing transmited, but at random intervals and without seeing the "transmitted tx_timestamp" or "finished" message in the terminal, the TX_FIFO doesn't stop. 
When running TX_Demo:
```
pi@raspberrypi:~/CC1101-master/examples/raspi $ sudo ./TX_Demo -v -a1 -r3 -i1000 -t5 -c1 -f434 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 3
Frequency: 2
Channel: 1
My_Addr: 1
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x01
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2
0x02 0x26 0x09 0xB6 0x00 0xEC 0x2C 0x19 0x11 0x51
0x00 0x59 0x7F 0x3C 0x81 0x3F 0x0B
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8
 #:0x01
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8
 #:0x02
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8
 #:0x03
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8
 #:0x04
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8
 #:0x05
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8
 #:0x06
[...]
```
And so on...

Any clues?
Thanks.

SpaceTeddy, 2018-09-06:
Do you have also a receiver, or TX only?

xluccianox, 2018-09-06:
I'm trying TX only, watching the spectrum in GQRX.

SpaceTeddy, 2018-09-07:
Well, with my SDR Hardware (rtl-sdr , lime-sdr), I can also not see a clean spectrum in Gqrx. I suggest that you try to use a RX as well to verify that the system is running well.

ramybt, 2018-10-26:
i am new to coding and still learning, i would like to use my cc1101 Transceiver to receive and send RF signals 315mhz-433mhz and maybe 915mhz, i have followed the instructions for this link [https://github.com/SpaceTeddy/CC1101/blob/master/README.md](url) and i think i did everything right, i have 433mhz door sensor and i tried to receive the signal of the door sensor to with my C1101 module on my RAPBERRY PI3 and  nothing happened, can anyone please tell me what is the problem, 
this is my pi here:

``root@raspberrypi:/home/pi/wiringPi# sudo ./RX_Demo -v -a3 -c1 -f433-915 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x0F
Version   : 0x0F
...done!

esdad697, 2018-12-10:
About a year later, this works! 

Replacing Raspberry Pi's and CC1101's did not help. OS upgrade did not fix with old version of this library (Reinstalled Raspbian on fresh SD, and enabled SPI. Wiring was already installed.) Tried latest version of this library, and it worked. So it seems like a it was a change to the library that fixed issue, but could have been a change I made a year ago to the source or something else. Glad it works.

Thanks for the help.

Here is receiver:
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x04
...done!
Mode: 3
Frequency: 2
Channel: 1
My_Addr: 3
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x03 
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0 
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2 
0x02 0x26 0x09 0xB6 0x04 0xEC 0x2B 0x19 0x11 0x41 
0x00 0x59 0x7F 0x3C 0x81 0x3F 0x0B 
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0 
RX_FIFO:0x06 0x03 0x01 0x00 0x00 0x0B 0xB8 | 0x49 0xB3 |
RSSI:-42 LQI:0x33 CRC:0x00 
TX_FIFO: 0x05 0x01 0x03 0x41 0x63 0x6B 
Ack_sent!
RX_FIFO:0x06 0x03 0x01 0x00 0x00 0x17 0x7A | 0x49 0xB3 |
RSSI:-42 LQI:0x33 CRC:0x00 
TX_FIFO: 0x05 0x01 0x03 0x41 0x63 0x6B 
Ack_sent!

closed 2018-12-10 by esdad697