SpaceTeddy/CC1101 #13: The example does not work following everything proposed in the readme from arduino to raspberry pi 3

ElPernotador, 2017-08-12:
Hello, we are trying to connect an Arduino Uno R3 with raspberri PI 3.

We installed in the raspberri the tx_demo and seems to work correctly, since we disconnected some pin on purpose and the example did not charge throwing error of module not found.

Then we load the arduino example, this runs until the line of the setup:

Cc1100.sidle (); // set to ILDE first

Exactly locks in the "while" of the next function

// ------------------------------- [end] -------------- ----------------------------
// | =========================== SPI Transmission ================ ============= |
Uint8_t CC1100 :: spi_putc (uint8_t data)
{
     SPDR = date;
   While (! (SPSR & (1 << SPIF))); // Warten bis Byte gesendet wurde
      Return SPDR;
}

We have checked the connections many times and we do not see any errors.

Any ideas?

Thank you!!!

SpaceTeddy, 2017-08-12:
hi ElPernotador,

- you want to connect a rPi 3 with an Arduino via a CC1101 module?
- you said, that the tx_demo of the rPi 3 seems to work correctly?
- you have trouble with the Arduino example, which stops at function cc1100.sidle ();?
- may you post the Arduino serial output?

BTW: I don't know your CC1101 module, but please be sure that this module can handle the 5V voltage level output of the Arduino. As I remember, the Arduino has 5V GPIO output, even you has connected VCC to 3.3V. The pure CC1101 IC is NOT 5V tolerant. This could be the issue.

regards,
Chris

ElPernotador, 2017-08-12:
Hello!

If we are trying to connect an arduino to a raspberry

In the raspberry seems to work correctly, attached initial log:

> $ ./TX_Demo -v -a1 -r3 -i1000 -c1 -m250 -f434
> CC1100 SW: Verbose option is set `'
> Raspberry CC1101 SPI Library test
> Init CC1100 ...
> Partnumber: 0x00
> Version: 0x14
> ... done!
> Mode: 4
> Frequency: 2
> Channel: 1
> My_Addr: 1
> Config Register:
> 0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x01
> 0x01 0x0B 0x00 0x10 0xB0 0x71 0x2D 0x3B 0x73 0xA0
> 0xF8 0x00 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2
> 0x02 0x26 0x09 0xB6 0x04 0xEC 0x29 0x17 0x11 0x45
> 0x00 0x59 0x7F 0x3C 0x81 0x3F 0x0B
> PaTable:
> 0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0
> TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8
>  #: 0x01
> TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE8
>  #: 0x02

And so on indefinitely

The log on the Arduino is:

> Init CC1100 ...
> Partnumber: 00
> Version: 14
> In EEPROM settings ...> 

And there are no more changes

As for your clarification of the 5v, we try this example:

[Https://github.com/veonik/arduino-cc1101/blob/master/examples/example.ino](Https://github.com/veonik/arduino-cc1101/blob/master/examples/example.ino)

Between 2 arduinos and works perfectly with similar connectors, but we are impossible to configure it to connect with a raspberry

THANK YOU

SpaceTeddy, 2017-08-12:
ok, you can read the Partnumber and Version of the chip, which means that your SPI interface is correct. But I can not see any EEPROM settings. Did you upload your personal eeprom data, which the Arduino compiler generates for you, to the Arduino board? I mean the eeprom python script, which is mentioned on the ReadMe site.

ElPernotador, 2017-08-12:
Really, thank you very much. The answer was before our eyes but we did not see it.

I work perfect.

We only have one doubt, why is it necessary to record this data in the EEPROM?

SpaceTeddy, 2017-08-12:
Happy that I could help you! 

The only reason I made it in this way is to safe flash memory. If you want to use only one modulation setting i.e. Gmsk 100kbit, you can hard code it in your main Arduino Sketch. It is up to you ;)

ElPernotador, 2017-08-12:
All clear! thanks a lot

closed 2017-08-12 by ElPernotador