SpaceTeddy/CC1101 #14: SPI1 not supported?

salmg, 2017-08-21:
Hello there,

Amazing work with this project.

I was trying to use your library with SPI1 instead of SPI0

[image]

I added the dtoverlay=spi1-3cs at /boot/config.txt
Also I changed the wiringPiSPISetup in the cc1100_raspi.cpp file for the "1" instead of the "0" channel and same changes in the wiringPiSPIDataRW method, but I got nothing:

$ sudo ./TX_Demo -v -a1 -r0 -i1000 -t5 -c1 -f315 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
no CC11xx found!

any ideas?

SpaceTeddy, 2017-08-21:
Hi salmg,

I didn't tried to change the SPI channel, but this CC1101 library uses wiringPi for the raspberryPi SPI communication. Please check the SPI documentation here: http://wiringpi.com/reference/spi-library/

Please modify the following functions to SPI channel 1 in the cc1100_raspi.cpp file:
- wiringPiSPISetup (0, ...) -> wiringPiSPISetup (1, ...)
- wiringPiSPIDataRW (0,...) -> wiringPiSPIDataRW (1,...)

I did not tested it, but please try it.

salmg, 2017-08-22:
Hello,

I tried different things but none worked.

So I decided to try the second SPI device as slave with this configuracion:
[image]

It is working now with two projects sharing the same SPI0. Thanks.

closed 2017-08-22 by salmg

SpaceTeddy, 2017-08-22:
Yes, this is the simplest solution if your speed requirements for two devices are ok.

salmg, 2017-08-25:
that's true! do you have an email where I can contact you?

SpaceTeddy, 2017-08-25:
spaceteddy1@gmx.net