SpaceTeddy/CC1101 #15: Arduino not reveicing anything

kyonmiriam, 2017-10-21:
While I try to use this lib to get my Arduinos communicate together, I follow the steps in the Readme.md, however,the RX_Demo shows following below
Partnumber:F1
Version:F1
...done
Mode:4
Frequency:2
Channel:1
My_Addr:3
Cfg_reg:
06 2E 80 07 57 43 3E 0E 45 03 
01 0B 00 10 B0 71 2D 3B 73 A0 
F8 00 07 0C 18 1D 1C C7 00 B2 
02 26 09 B6 04 EA 0A 00 11 41 
00 59 7F 3A 81 3F 0B 
PaTable:
6C 1C 06 3A 51 85 C8 C0 

and nothing more. The TX_Demo shows like this 

Init CC1100...
Partnumber:00
Version:14
...done
Mode:4
Frequency:2
Channel:1
My_Addr:1
Cfg_reg:
06 2E 80 07 57 43 3E 0E 45 01 
01 0B 00 10 B0 71 2D 3B 73 A0 
F8 00 07 0C 18 1D 1C C7 00 B2 
02 26 09 B6 04 EA 0A 00 11 49 
00 59 7F 3C 81 3F 0B 
PaTable:
6C 1C 06 3A 51 85 C8 C0 
CC1101 TX Demo
TX_FIFO:060301000003E8
 #:00
TX_FIFO:060301000003E8
 #:01
tx_time: 511ms
TX_FIFO:060301000009CF
 #:00
TX_FIFO:060301000009CF

So is it all right?

SpaceTeddy, 2017-10-21:
dear kyonmiriam,

for your receiver the partnumber and Version number looks not correct. it should not 0x1F.
Anyhow, I seems that the cc1101 config registers are written correct. 
May you please verify that GDO2 pin of the CC1101 is connected to Pin 3 of your Arduino?

thx

kyonmiriam, 2017-10-22:
Hi,
I do verify my connection and surely that GDO2 pin of the CC1101 is connected to Pin 3 of Arduino(Not pin A3). 
However, the RX_Dome still shows messages like this

Partnumber:F1
Version:F1
...done
Mode:4
Frequency:2
Channel:1
My_Addr:3
Cfg_reg:
70 2E FC 47 57 43 2E C0 45 03 
61 39 00 10 B0 71 2D 3B 73 A0 
20 00 1B 0C 18 1D 1C C7 00 B2 
02 FF 08 B6 37 EC 2B 39 11 4A 
00 B7 7F 80 88 31 0B 
PaTable:
3E 3E 3E 3E 3E 3E 3E 3E 

Then I rewrote the EEPROM of the Arduino with the RX_Demo sketch, It still behaves like this 
Partnumber:00
Version:F1
...done
Mode:4
Frequency:2
Channel:1
My_Addr:3
Cfg_reg:
C7 C7 E0 3A B8 23 F0 77 26 3E 
7E E8 3B 62 90 23 1F D4 08 55 
A0 57 08 13 42 F3 FF F8 36 A0 
1E D0 1F D0 12 3F 12 2F C0 BB 
C1 17 FF 00 3F B0 3D 
PaTable:
3E 3E 3E 3E 3E 3E 3E 3E 

Is there anything incorrect I haven't got noticed?

SpaceTeddy, 2017-10-22:
Hi,
No. The config register and the PA table are not correct written/read (SPI problem) or the eeprom content is wrong. It can also the case that the Cc1101 itself is damaged or pre damaged. You should verify that the correct eeprom content is written successfully to the Arduino. Then cross check that the cc1101 is connected to 3.3v and not to 5v. 
I have double checked with my setup that the library is still working. Because I’m not an experienced programmer, I can happen that I’m doing crazy mistakes ;))

kyonmiriam, 2017-10-22:
Emmm...thx a lot.
I have change another new CC1101.
When I compile the eeprom_write.ino with the eep array  sketch & upload into to my Arduino  I noticed that the Arduino Serial console shows massage like that
Settings:
ISM Mode: 3
Modulation Mode: 4
My Addr: 3
Channel: 1

*------ Menu -------*
*press 'w' for write*
*press 'r' for read *
*press 'b' for blank*
=====================

However in the RX_Dome the setting is 
`cc1100.sidle();                          
  cc1100.set_mode(0x04);                 
  cc1100.set_ISM(0x02);                    
  cc1100.set_channel(0x01);               
  cc1100.set_output_power_level(0);        
  cc1100.set_myaddr(0x03);                 
`
Notice that ISM band is 0x02 not 0x03, what's wrong?

SpaceTeddy, 2017-10-22:
This is no problem. Important is, that you use the python script to generate the eeprom array out of the eep file from the RX or TX_Demo sketch. The settings which you see from the eeprom sketch are only default values. After flashing the Demo sketches, the corrects setting are used.
If you have flashed the correct eep file to your arduino, your problem is likely your cable connection or power supply.

kyonmiriam, 2017-10-23:
All right, after I've changed a shorter cable, it works!
Thanks a lot. But how could I send a custom lens or char via this demo?

SpaceTeddy, 2017-10-23:
Great, you got it!
If you want to send custom chars, just fill up the Tx_fifo Array with your data.
Please note that Tx_fido[0-2] is used for protocol header. You can use [3-63] for your data.

closed 2017-10-24 by SpaceTeddy