SpaceTeddy/CC1101 #16: Trying to receive signal from CCdebugger and smartRF studio

Bob1237, 2017-11-30:
Hello

FIrst let me tell you i am a complete noob to raspberry and cc1101 , so i don't know a lot about it.
But i was wondering if it was possible to use your code to receive a signal with the raspberry pi and cc1101 send by smart RF studio and the ccdebugger. Because I have followed your readme file and did everthing like you said , but i don't receive anything , and  i am using the same frequentcy , modulation ,... 
My problem is similar to the one were you talk about inthe issues: Not Receiving Anything #11
but nothing he did there works for me.

if you know anything about htis let me know

Thanks 

BoB1237

SpaceTeddy, 2017-11-30:
Hi Bob, 
thanks to try my library. 

May you please post some terminal  output logs of the RX_Demo and TX_Demo? In addition, detailed pictures of your hardware Setup are helpful!

Thank you,

Kind regards,
Chris

Bob1237, 2017-12-07:
hello

this is my terminal output : 
[image]

and here are picture of the connection but i don't think there something wrong whith that because i connect it exactly the same like you. (sorry for the bad picture) 
[image]
[image]
[image]

first one is from the transmitting end

Sorry for the late rasponds

Bob

SpaceTeddy, 2017-12-07:
Dear Bob,

your register settings looks not correct and you used non valid command line parameters.
May you please follow the example command line parameters?
try: 
sudo ./RX_Demo -v -a3 -c1 -f434 -m100 (receiver address 3; channel 1; frequency 434; modulation 100kbit GMSK)

I've never used this CCdebugger. Unfortunatly I can not help you.

SpaceTeddy, 2018-01-04:
did you try again to keep it running?

closed 2019-01-14 by SpaceTeddy