SpaceTeddy/CC1101 #17: Suggested improvements to your example

Phil-666, 2017-12-29:
Hi,
I'm not very experienced with Arduino code and so this took me a little longer than normal. I've just managed to get your example code working and thought I'd give you feedback on the extra steps I had to perform to make it work.

My environment was Arduino 1.6.9 on a Windows platform, and a Rasberry Pi Zero W.

1) Put each .ino file into a directory of the same name. Otherwise Arduino complains, but then does it for you.
2) Using Windows File Explorer, create a new folder within your existing Arduino\libraries folder, and place the cc1100_arduino.cpp & cc1100_arduino.h files in there. This allows the use of the '#include <>' contruct
    e.g. C:\Users\<YOUR_USERNAME>\Documents\Arduino\libraries\CC1101
3) Within Tx_demo.ino, replace #include <cc1100.h> with #include <cc1100_arduino.h> (or rename the library .cpp & .h files)
4) To locate the 'Tx_demo.ino.eep' file look in C:\Users\<YOUR_USERNAME>\AppData\Local\Temp\build*.tmp
This file is typically less than 1KB in size, do not use the .hex file like I did :)
5) By default the TX is configured to
    Mode:4 (=MSK_250_kb)
    Frequency:2 (=433MHz)
    Channel:1
    My_Addr:1
Note, within the TX code, the RX address is set to 0x03 on line 70 of Tx_demo.ino (This could be configured at the top and displayed in the settings)

Therefore the RX needs to run as:
sudo ./RX_Demo -v -a3 -c1 -f434 -m250

After fixing the above issues the code worked a treat, many thanks,
Phil

SpaceTeddy, 2018-01-04:
Dear Phil,

thank you that you play around with my library and i'm happy that you get it running.
Regarding your comments... so, you mean, that I should improve the parts in the ReadMe file to make it more clear how to compile the Arduino sketch?

https://github.com/SpaceTeddy/CC1101#arduino-specific

regards,
Chris

Phil-666, 2018-01-05:
Hi Chris,
yes that is what I meant, just to add a few tips to other newbies. Maybe you could update the code for my point (3).

Bye the way, I orginally compiled the TX_demo code using Arduino 1.6.9 and it worked fine. However when I compiled it with the newer Arduino 1.8.5 it ran for a bit then crashed during the cc1101.transmit() function. I've not had time to debug it myself, but again a warning for others.

Phil

SpaceTeddy, 2018-01-05:
Dear Phil,
Yes you’re right. If you change the Arduino version, you most likely change also the GCC Compiler Version. This has an influence to the the eeprom file, because the compiler chooses the eeprom position of the parameters. Please compare the different eep files and you will see tiny differences. I wrote this in the Readme as well, but probably not strong enough. If I have a bit time, I will update the remark. Unfortunately time is currently very limited :(

Kind regards,

Chris

closed 2019-01-14 by SpaceTeddy