SpaceTeddy/CC1101 #21: Receiving OOK packet on 433,950 MHz

dudator, 2018-05-19:
Hello

First I will thank you for your awesome code. I really appreciate it!

My problem is that I'm trying to receive a packet that is send on 433,950MHz with OOK. If I send a packet with OOK, it works as I can review it with a RTL-SDR Dongle. I know that OOK is not available in the command line options but I enabled it in extending you switch-case statement with a additional case and sets the cc1100_mod_select value to 6. With my dongle I was able to determine the channel for 433,950 to channel 0.

For my testcase I use my car-key that sends out OOK packets at 433,950. But I'm not able to determine the correct address for my program. I don't know if an address is needed in this case? Can I set my CC1101 to some sort of promiscuous mode to receive all signals that will be sent on this frequency?

My command looks like(m1000 is the new switch-case value for OOK mode):

`sudo ./RX_Demo -v -a3 -c0 -f434 -m1000
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 6
Frequency: 2
Channel: 0
My_Addr: 3
Config Register:
0x06 0x2E 0x06 0x47 0x57 0x43 0xFF 0x04 0x05 0x03
0x00 0x06 0x00 0x10 0xB0 0x71 0x87 0x83 0x3B 0x22
0xF8 0x15 0x07 0x30 0x18 0x14 0x6C 0x07 0x00 0x92
0x87 0x6B 0xFB 0x56 0x04 0xEF 0x2B 0x14 0x1F 0x41
0x00 0x59 0x7F 0x3F 0x81 0x35 0x09
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0`

Thank you very much!

esdad697, 2018-12-10:
Would -m6 set mode to 6? 

I tried transmitter and receiver with -m6, but messages were not received:
- sudo ./TX_Demo -v -a1 -r3 -c1 -f434 -m6 -i3000
- sudo ./RX_Demo -v -a3 -c1 -f434 -m6

Is this the right way to select OOK/ASK modulation? Has anyone gotten this to work?

SpaceTeddy, 2019-01-14:
Mode 6 is available now.
Thanks to kings-way https://github.com/kings-way/CC1101/commit/5285d19884d7721bc9bfab6f474763a1545eaa80#diff-13e75c32f716596c59eab4ce1023fd75

closed 2019-01-14 by SpaceTeddy