SpaceTeddy/CC1101 #23: CC1101 raspberry and arduino

ahmedgh1995, 2018-08-14:
i wanted to run ur code on an arduino as a sender and a raspberry as a reciever but althought both code seems to be working just fine ,there is no paquet recieved in the raspberry .

SpaceTeddy, 2018-08-14:
Hi, thank you to play around with my code, but may you please give me some more detailed information, how did you used it. Did you play with the examples, and which command did you use?
Any logs or screenshots would be helpful to analyse your issue.

Thx, 
Chris

ahmedgh1995, 2018-08-14:
hello

here is a screen shot of my two codes running on both arduino and raspberry

also u can find attached my code

[cid:0a346bf7-6eb3-4129-9716-21ed1ffbde66]

'                     CC1100 Raspberry Pi Library
'                     ---------------------------
'
'
'
'
'
'  module contains helper code from other people. Thx for that
'-----------------------------------------------------------------------------*/
#include "cc1100_raspi.h"
#include <stdio.h>
#include <stdint.h>
#include <string.h>
#include <wiringPi.h>
#include <wiringPiSPI.h>

extern uint8_t cc1100_debug;
//-------------------[global default settings 868 Mhz]---------------------------------

static uint8_t cc1100_GFSK_1_2_kb[CFG_REGISTER] = {
                    0x07,  // IOCFG2        GDO2 Output Pin Configuration
                    0x2E,  // IOCFG1        GDO1 Output Pin Configuration
                    0x80,  // IOCFG0        GDO0 Output Pin Configuration
                    0x07,  // FIFOTHR       RX FIFO and TX FIFO Thresholds
                    0x57,  // SYNC1         Sync Word, High Byte
                    0x43,  // SYNC0         Sync Word, Low Byte
                    0x3E,  // PKTLEN        Packet Length
                    0x0E,  // PKTCTRL1      Packet Automation Control
                    0x45,  // PKTCTRL0      Packet Automation Control
                    0xFF,  // ADDR          Device Address
                    0x00,  // CHANNR        Channel Number
                    0x08,  // FSCTRL1       Frequency Synthesizer Control
                    0x00,  // FSCTRL0       Frequency Synthesizer Control
                    0x21,  // FREQ2         Frequency Control Word, High Byte
                    0x65,  // FREQ1         Frequency Control Word, Middle Byte
                    0x6A,  // FREQ0         Frequency Control Word, Low Byte
                    0xF5,  // MDMCFG4       Modem Configuration
                    0x83,  // MDMCFG3       Modem Configuration
                    0x13,  // MDMCFG2       Modem Configuration
                    0xA0,  // MDMCFG1       Modem Configuration
                    0xF8,  // MDMCFG0       Modem Configuration
                    0x15,  // DEVIATN       Modem Deviation Setting
                    0x07,  // MCSM2         Main Radio Control State Machine Configuration
                    0x0C,  // MCSM1         Main Radio Control State Machine Configuration
                    0x18,  // MCSM0         Main Radio Control State Machine Configuration
                    0x16,  // FOCCFG        Frequency Offset Compensation Configuration
                    0x6C,  // BSCFG         Bit Synchronization Configuration
                    0x03,  // AGCCTRL2      AGC Control
                    0x40,  // AGCCTRL1      AGC Control
                    0x91,  // AGCCTRL0      AGC Control
                    0x02,  // WOREVT1       High Byte Event0 Timeout
                    0x26,  // WOREVT0       Low Byte Event0 Timeout
                    0x09,  // WORCTRL       Wake On Radio Control
                    0x56,  // FREND1        Front End RX Configuration
                    0x17,  // FREND0        Front End TX Configuration
                    0xA9,  // FSCAL3        Frequency Synthesizer Calibration
                    0x0A,  // FSCAL2        Frequency Synthesizer Calibration
                    0x00,  // FSCAL1        Frequency Synthesizer Calibration
                    0x11,  // FSCAL0        Frequency Synthesizer Calibration
                    0x41,  // RCCTRL1       RC Oscillator Configuration
                    0x00,  // RCCTRL0       RC Oscillator Configuration
                    0x59,  // FSTEST        Frequency Synthesizer Calibration Control,
[... 1340 lines cut ...]

        void rx_fifo_erase(uint8_t *rxbuffer);
        void tx_fifo_erase(uint8_t *txbuffer);

        uint8_t sent_packet(uint8_t my_addr, uint8_t rx_addr, uint8_t *txbuffer, uint8_t pktlen, uint8_t tx_retries);
        void sent_acknolage(uint8_t my_addr, uint8_t tx_addr);

        uint8_t check_acknolage(uint8_t *rxbuffer, uint8_t pktlen, uint8_t sender, uint8_t my_addr);

        int8_t rssi_convert(uint8_t Rssi);
        uint8_t check_crc(uint8_t lqi);
        uint8_t lqi_convert(uint8_t lqi);
        uint8_t get_temp(uint8_t *ptemp_Arr);

        void set_myaddr(uint8_t addr);
        void set_channel(uint8_t channel);
        void set_ISM(uint8_t ism_freq);
        void set_mode(uint8_t mode);
        void set_output_power_level(int8_t dbm);
        void set_patable(uint8_t *patable_arr);
        void set_fec(uint8_t cfg);
        void set_data_whitening(uint8_t cfg);
        void set_modulation_type(uint8_t cfg);
        void set_preamble_len(uint8_t cfg);
        void set_manchaster_encoding(u

SpaceTeddy, 2018-08-14:
Hi, sry I can not see the screenshots.
May you please copy/ Paste the output of the serial monitor of your Arduino and the output of you raspi terminal?! Thx

ahmedgh1995, 2018-08-14:
sorry for that,is this better ?

SpaceTeddy, 2018-08-14:
no, sry.

copy paste the output as text, not as screenshot. thx

ahmedgh1995, 2018-08-14:
arduino outpu :
********************************
Init CC1100...
Partnumber:00
Version:14
3318...done
8
Mode:3
Frequency:3
Channel:1
My_Addr:8
Cfg_reg:
07 2E 80 07 57 43 3E 0E 45 08
01 08 00 21 65 6A 5B F8 13 A0
F8 47 07 0C 18 1D 1C C7 00 B2
02 26 09 B6 04 EA 0A 00 11 41
00 59 7F 3C 81 3F 0B
PaTable:
03 17 1D 26 50 86 CD C0
CC1101 TX Demo
packet sent :
TX_FIFO:060308E0100804

raspberry output :

***********************************

pi@raspberrypi:/home/cc1101 $ ./RX_Demo -v -a3 -c1 -f868 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 3
Frequency: 3
Channel: 1
My_Addr: 3
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x03
0x01 0x08 0x00 0x21 0x65 0x6A 0x5B 0xF8 0x13 0xA0
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2
0x02 0x26 0x09 0xB6 0x04 0xEC 0x29 0x1B 0x11 0x51
0x00 0x59 0x7F 0x3C 0x81 0x3F 0x0B
PaTable:
0x03 0x17 0x1D 0x26 0x50 0x86 0xCD 0xC0

________________________________
De : Ahmed BEN SOLTANE
Envoyé : mardi 14 août 2018 18:10:56
À : SpaceTeddy/CC1101; SpaceTeddy/CC1101
Cc : Author
Objet : RE: [SpaceTeddy/CC1101] CC1101 raspberry and arduino (#23)

sorry for that,is this better ?

ahmedgh1995, 2018-08-14:
> ok, which cc1101 modules did you use? are they made for 868MHz? or did you use modules for 434MHz?

My modules are 868 mhz

Télécharger Outlook pour Android<https://aka.ms/ghei36>

SpaceTeddy, 2018-08-15:
well, which example do you use for the tests? The standard RX_demo, TX_demo version, or the more sensitive WOR version? Have you done any modification to the code (instead of setting the modes)?

ahmedgh1995, 2018-08-15:
./RX_Demo -v -a3 -c1 -f868 -m100

Télécharger Outlook pour Android<https://aka.ms/ghei36>

send a picture of your test setup, please!