SpaceTeddy/CC1101 #31: Unable to compare/process received data

Siem-H, 2019-02-10:
I am trying to modify the RX_Demo to trigger a script based on the received data, but I am unable to compare the received data to anything. 
I don't know how to compare the Rx_fifo buffer to a pre-set value. 
I want to make a simple if comparison, e.g. 
> if (Rx_fifo == 0x06 0x02 0x01 0x01 0x01 0x01 0x01)

But that is clearly not the solution.
Can someone help me?

SpaceTeddy, 2019-02-11:
Hi,
basically there are different ways to compare arrays.
you may try the function 'memcmp' function.

your compare array:
int trigger[] = {0x06, 0x02, 0x01, 0x01, 0x01, 0x01, 0x01}; 

copmare with Rx_fifo:
int n=0;
n=memcmp ( trigger, Rx_fifo, sizeof(trigger) );

if the variable n is == 0, then the campare matches.

try it

Siem-H, 2019-02-11:
It keeps not returning 0, even though it seems to match.
Does the ' | 0xF5 0xAB | ' behind the RX_FIFO matter? If it does, how do I add it to the trigger / exclude it from the compare?
I also tried comparing with std::search but that kept returning 0 even though it didn't match, I even tried to change the trigger to only contain the data (0x01, 0x01, 0x01, 0x01), but that didn't work either.

SpaceTeddy, 2019-02-12:
yes, the last 2 bytes in the Rx_fifo are the RSSI and LQI value. These will change almost every time. Therefore you should only compare your real payload. But IMHO this is covered with `sizeof(trigger)`. That means that only the count of bytes in trigger[] is compared.

Siem-H, 2019-02-12:
// Global variable
`int trigger[] = {0x06, 0x02, 0x01, 0x01, 0x01, 0x01, 0x01};`

//...... (just stock stuff between these two)

        for (;;) {
                delay(1);                            //delay to reduce system load
                if (cc1100.packet_available())           //checks if a packed is available
                {
                  cc1100.get_payload(Rx_fifo, pktlen, rx_addr, sender, rssi_dbm, lqi); //stores the payload data to Rx_fifo
                  int n=0;
                  printf("comparing ... \r\n\r\n");
                  n=memcmp (trigger, Rx_fifo, sizeof(trigger));
                  // n=std::search(Rx_fifo, Rx_fifo + sizeof(Rx_fifo), trigger, trigger + sizeof(trigger)) != (Rx_fifo + sizeof(Rx_fifo));
                  // also tried this with int trigger[] = {0x01, 0x01, 0x01, 0x01};
                  if (n == 0)
                  {
                    printf("YES \r\n\r\n");
                  }else{
                    printf("NO \r\n\r\n");
                  }
                }
        }

SpaceTeddy, 2019-02-12:
Which are your Sender & Receiver addresses?

Siem-H, 2019-02-12:
Sender is 1, received is 2

SpaceTeddy, 2019-02-14:
you can also do:

`int compareArrays(int a[], int b[], int n)
{
  for (int i=0; i<n; ++i)
  {
      if (a[i] != b[i])
      {
          return -1;
          break;
      }
  }
  return 0;
}
`

Siem-H, 2019-02-15:
What is 'n' in this case?

adminy, 2024-08-01:
probably `sizeof(trigger)`, so `compareArrays(Rx_fifo, trigger, sizeof(trigger))`. I've made this simpler by printing out program output and parsing that output in another program.