SpaceTeddy/CC1101 #32: MSP430 MCU support is added by me

abhra0897, 2019-02-26:
Hi **SpaceTeddy**, thank you for your awesome library for interfacing **CC1101** with **AVR (Arduino)** and **Raspberry Pi**. 

However, I have recently added support for **Texas Instrument's MSP430** MCUs. It's tested and works properly. My library is also completely compatible with yours and for that reason, communication between **MSP430** and **AVR** is possible (Tested).

If you add my library with your repository, that will be helpful to many others, or you can just mention it in your readme - whatever you think is the best.

- My library's link: https://github.com/abhra0897/msp430_cc1101_energia_v2
- Testing video link: https://www.youtube.com/watch?v=tSoDvinVhsI&t=46s

Best regards,
Avra

SpaceTeddy, 2019-02-26:
Dear Avra,
Cool, that you added MSP430 support for this library. Awesome!!!
Unfortunately I don’t have any time to support this library in the future, there I think it is the best to add an link to your library in my readme. If you want to add further features, then it is easier for you to add these items. 
Please give me some days to add it

Thank you to bring this project further.

Chris

abhra0897, 2019-02-26:
Thank you for your reply. Add the link when you are free, no problem.
I'm really grateful to you for this library as it helped us tremendously in a cubesat project.
Now it supports 3 major platforms- AVR, MSP430 and Raspberry pi. Next I'll try to add STM32 support in my spare time.