SpaceTeddy/CC1101 #33: How were original register settings determined?

esdad697, 2019-02-28:
Did you use SmartRF Studio or some other reference?

SpaceTeddy, 2019-02-28:
Hi esdad697,

yes, I used SmartRF Studio to get the basic settings up and running. If you want to use special RF features, I recommend that you use SmartRF Studio to change the parameters. There are not so many parameters that need to be changed, so it is also possible to calculate the values manually. Based on the data sheet of course.

esdad697, 2019-03-01:
In SmartRF Studio, I take the GFSK default and change the frequency to 345. I can send and receive.
Back in SmartRF, the modulation is changed to ASK. Nothing is received. I copied the patable too. Should i copy the patable? 

TI said studio is not optimized for ASK, and suggested: [http://www.ti.com/lit/an/swra215e/swra215e.pdf](url)
I can follow most of it, so will give it a try.

SpaceTeddy, 2019-03-01:
well,
what do you would like to do?
I guess in your case I would like to go step by step. If you want to use ASK/OOK with a different Frequency, set the frequency to 345MHz on both sides and test. If it is successful, modify the second parameter and try again. Means, take your working parameter set and change as less a possible and go on step by step.

esdad697, 2019-03-02:
I can't get ASK to work. I tried following TI's design note. It was not too complicated, but did not help.
I'll take it to TI: [(https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/p/777909/2878685#2878685)]

Maybe CC1101 is not a good choice for ASK. I was also planning on trying async mode to read older signal.

SpaceTeddy, 2019-03-03:
You can try my demo with ASK / OOK.

closed 2019-03-03 by SpaceTeddy

SinanAkkoyun, 2019-10-25:
Hey SpaceTeddy,
I am hardly trying for hours to make myself familiar with SmartRF, but I miserably fail.
Can you please help me with setting the CC1101 in async mode, using GDO2 as the serial data output at 433MHz and 868MHz or at least tell me how to get there?
I should just be able to receive and send the data I got from GDO2 to GDO0 over UART, right?
Thanks in advance :)

DerrowBond, 2020-11-25:
These are my register settings for 433 Mhz ASK/OOK using a 26 Mhz quartz in async mode for keeloq receiving.
Received signal will be on GD02.
Settings are optimized for high sensitivity.

		0x0D,  // IOCFG2              GDO2 Output Pin Configuration
		0x2E,  // IOCFG1              GDO1 Output Pin Configuration
		0x3f,  // IOCFG0              GDO0 Output Pin Configuration
		0x47,  // FIFOTHR             RX FIFO and TX FIFO Thresholds
		0xD3,  // SYNC1               Sync Word, High Byte
		0x91,  // SYNC0               Sync Word, Low Byte
		0xFF,  // PKTLEN              Packet Length
		0x04,  // PKTCTRL1            Packet Automation Control
		0x32,  // PKTCTRL0            Packet Automation Control
		0x00,  // ADDR                Device Address
		0x00,  // CHANNR              Channel Number
		0x06,  // FSCTRL1             Frequency Synthesizer Control
		0x00,  // FSCTRL0             Frequency Synthesizer Control
		0x10,  // FREQ2               Frequency Control Word, High Byte
		0xB0,  // FREQ1               Frequency Control Word, Middle Byte
		0x71,  // FREQ0               Frequency Control Word, Low Byte
		0xA7,  // MDMCFG4             Modem Configuration
		0x32,  // MDMCFG3             Modem Configuration
		0x30,  // MDMCFG2             Modem Configuration
		0x22,  // MDMCFG1             Modem Configuration
		0xF8,  // MDMCFG0             Modem Configuration
		0x44,  // DEVIATN             Modem Deviation Setting
		0x07,  // MCSM2               Main Radio Control State Machine Configuration
		0x30,  // MCSM1               Main Radio Control State Machine Configuration
		0x18,  // MCSM0               Main Radio Control State Machine Configuration
		0x1D,  // FOCCFG              Frequency Offset Compensation Configuration
		0x1C,  // BSCFG               Bit Synchronization Configuration
		0x04,  // AGCCTRL2            AGC Control
		0x00,  // AGCCTRL1            AGC Control
		0x92,  // AGCCTRL0            AGC Control
		0x87,  // WOREVT1             High Byte Event0 Timeout
		0x6B,  // WOREVT0             Low Byte Event0 Timeout
		0xFB,  // WORCTRL             Wake On Radio Control
		0xB6,  // FREND1              Front End RX Configuration
		0x11,  // FREND0              Front End TX Configuration
		0xE9,  // FSCAL3              Frequency Synthesizer Calibration
		0x2A,  // FSCAL2              Frequency Synthesizer Calibration
		0x00,  // FSCAL1              Frequency Synthesizer Calibration
		0x1F,  // FSCAL0              Frequency Synthesizer Calibration
		0x41,  // RCCTRL1             RC Oscillator Configuration
		0x00,  // RCCTRL0             RC Oscillator Configuration
		0x59,  // FSTEST              Frequency Synthesizer Calibration Control
		0x7F,  // PTEST               Production Test
		0x3F,  // AGCTEST             AGC Test
		0x81,  // TEST2               Various Test Settings
		0x35,  // TEST1               Various Test Settings
		0x09,  // TEST0               Various Test Settings

SinanAkkoyun, 2020-11-25:
Oh @DerrowBond thank you so much! Even though this is from long ago, it is as much of use to me right now!!!

How did you learn to get familiar with SmarRF and tune the settings for your need?

DerrowBond, 2020-11-25:
I am glad you can use it :)
Well I started with CC1101 about a week ago, and studied many papers to finally get it working to my needs.
A big help has been two posts on the Ti Forum:
https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/t/629112?CC1101-async-vs-sync-mode-how-decode-data-from-raw
https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/t/148027

I am using the CC1101 as receiver for Keeloq signals sent in AM and FM, and use a Raspberry Pi to decode the Keeloq Signal.

SinanAkkoyun, 2020-11-25:
Yeah cool, thanks for the link! I also made my stuff work back then, but only on that frequency with these settings xD

I wish you the best on your projects!