SpaceTeddy/CC1101 #34: Stuck in Tx - MARCSTATE 0x13

jaydip-kathiriya, 2019-03-09:
Hi,

I have ported your library for use in STM32F4. I have changed the library from C++ to C.

I am facing a very peculiar problem. I can receive everything. (Broadcast as well as Address specific transmissions), but I am unable to send an ack or send any kind of data to any device.

The module is stuck in MARCSTATE 0x13 (Tx) and never comes out of Tx.

Things I have checked:
- Verified that the TXFIFO is being sent the correct data
- Verified that data received is correct. (GDO2 is working properly for Rx)
- Changed the implementation of STX/SRX/SPWD/SIDLE to CS enable. Even then, Rx works, Tx doesn't.
- Verified that TXFIFO is being flushed before sending any data

Any help you can give will be much appreciated.

Thanks,
Jaydip

SpaceTeddy, 2019-03-13:
Hi Jaydip,
I never saw this hang up while TX. 
The only thing I could imagine is, that the TX configuration settings are not made that the CC1101 will not go back to IDLE after sending out the package. My implementation of transmit data relays on that the chip goes back to IDLE after a STX strobe command.
May you please check register 0x17: MCSM1– Main Radio Control State Machine Configuration
page 81 of CC1101 datasheet.

thx

jaydip-kathiriya, 2019-03-22:
Hello SpaceTeddy,

First of all thank you very much for your response but MCSM1 register value
as per the datasheet to come back into idle state after transmission.

But still it is not coming.

If you have some time i will make the setup and give you anydesk or team
viewer for live debug session. It may make you more clear about the problem.

As i have mentioned earlier that it is receiving properly but when again it
went for ACK it will be hang.

It will be your great great support if you will look into my code in debug
mode.

Thanks again.

*Jaydip Kathiriya (Founder)*

*The Techno Valley*

*A1-504, Shree Residency,*
*Ram Chowk, Sudama Chowk Road,*
*Mota Varachha,*
*Surat - 394101*
*+91 8866537049*