SpaceTeddy/CC1101 #35: Can I use this libary to sniff 868Mhz protocol with CC1101

guitar9, 2019-03-16:
Can I use this libary to sniff 868Mhz protocol with CC1101 ?
I want to reverse engineering the protcol of a MAX! Thermostat. 
I have a arduino mega 2560. I dont know how to do this. Is there a instruction how to use this libary and how to do the wiring?

This is my CC1101:
[image]

SpaceTeddy, 2019-03-16:
i'm not sure if it is possible to sniff signals. with my demo settings it is not possible, because the cc1101 is listening to dedicated addresses and need a specified packet length. 
But, as I remember right, the CUL Stick, which is used for the fhem home automation system can already listen to MAX! and has also a cc1101 on board.

esdad697, 2019-03-16:
No, it is not a sniffer. If you knew the protocol, this library might
provide a starting point to read it.
Often you can find the protocol on the web. Also, RTL-SDR can be used to
capture the signal. You could then view the signal and figure out the
encoding.