SpaceTeddy/CC1101 #36: Will the TX_Demo and RX_Demo work on one raspberry pi at the same time?

ole1986, 2019-03-19:
While I am struggling to get a signal I was asking myself if its even possible to use both commands, the TX_Demo and RX_Demo at the same time on a single raspberry pi (v1 Mod B+)

Thank you in advanced for any response

SpaceTeddy, 2019-03-21:
well, as I remember right, the PI has two SPI chip select pins, means you can theoretically use two SPI devices with one pi. But I guess in reality it needs effort do write a driver, which checks that the bus is free. If you want to transfer many bytes, you have to take care about the timings, therefore I guess that it is more effective to setup a second rPi.

ole1986, 2019-03-21:
@SpaceTeddy thank you for your answer. Highly appreciated

closed 2019-03-21 by ole1986