SpaceTeddy/CC1101 #39: Noob problems

CarpeEspion666, 2019-05-07:
Hello,
I did clone the project to my rpi and read your readme file. But now, I'm just blocked. I understood your code could help me for my project in node red. But i just don't know how to start the code know that I have the file on my pi.

Thanks for those who could help me or link me a tutorial, I searched for hours and didn't find anything.

SpaceTeddy, 2019-05-07:
Did you read it from here?
https://github.com/SpaceTeddy/CC1101/blob/master/README.md#raspberry-pi

CarpeEspion666, 2019-05-07:
Thank you for this quick answer. I did follow many tutorial and always got blocked by different issue. When i compile and start with those step, I just get 

CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
no CC11xx found!

is that an hardware problem? wiring?

testbughub, 2019-07-26:
Fellow noob here, having the exact same issue.
Hoping for any kind of help!

SpaceTeddy, 2019-07-26:
OK, which Hardware do you use? rPi or arduino? please be so kind and take a picture of your full test setup. Probably we can find the issue.

testbughub, 2019-07-26:
Never mind, I thought I had double checked the wiring, but after a tripple check I noticed one cable was on the wrong pin.
Thanks anyway, and for the quick reply!

SpaceTeddy, 2019-07-26:
👍

closed 2019-07-26 by SpaceTeddy