SpaceTeddy/CC1101 #45: CC1101 detected, but demo transmit fails

JYouren, 2020-06-01:
Hi,

I have compiled the TX demo as is, and attempted to run it. I get the following output.

pi@raspberrypi:~/CC1101/examples/raspi $ sudo ./TX_Demo -v -a1 -r3 -i1000 -t5 -c1 -f434 -m100
CC1100 SW: Verbose option is set `'
Raspberry CC1101 SPI Library test
Init CC1100...
Partnumber: 0x00
Version   : 0x14
...done!
Mode: 3
Frequency: 2
Channel: 1
My_Addr: 1
Config Register:
0x07 0x2E 0x80 0x07 0x57 0x43 0x3E 0x0E 0x45 0x01 
0x01 0x08 0x00 0x10 0xB0 0x71 0x5B 0xF8 0x13 0xA0 
0xF8 0x47 0x07 0x0C 0x18 0x1D 0x1C 0xC7 0x00 0xB2 
0x02 0x26 0x09 0xB6 0x04 0xEC 0x0B 0x3D 0x11 0x4A 
0x40 0x59 0x7F 0x3C 0x81 0x3F 0x0B 
PaTable:
0x6C 0x1C 0x06 0x3A 0x51 0x85 0xC8 0xC0 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE9 
 #:0x01 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE9 
 #:0x02 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE9 
 #:0x03 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE9 
 #:0x04 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE9 
 #:0x05 
TX_FIFO: 0x06 0x03 0x01 0x00 0x00 0x03 0xE9 
 #:0x06 
....

It seems like everything is setting up correctly, but the transmit packets keep failing the 6 retries

Is there anything obvious I should check?

Device is TI-CC1101 connected to Raspberry PI 2B via SPI pins.

Thanks!

JYouren, 2020-06-03:
It looks like transmit is actually working, but the transmit code in this library is looking for a response back. I am performing a fire-and-forget type scenario, so this is not relevant; I will continue my investigation.

SpaceTeddy, 2020-06-03:
hi,
long time ago, that I looked into this repo, but you can try to send the BROADCAST_ADDRESS. If you do so, the sender should not wait for an acknowledge. 

```
if(rx_addr == BROADCAST_ADDRESS){                       //no wait acknowledge if sent to broadcast address or tx_retries = 0
            return TRUE; 
```

JYouren, 2020-06-03:
Ah yes, very nice, thank you. Will check this.