SpaceTeddy/CC1101 #51: Receive raw data for replay

Ruegg, 2021-07-27:
Hi,

Is there anyway I can receive just the raw data and replay it later? This means I don't care for being able to understand the data format, modulation, or anything of the sort in my use case. I purely wish to be able to detect that some sort of RF is occurring, and replay it later.

I'm not too familiar with RF overall, but I tried to read the  CC1101 documentation by Texas instruments, and disable all the filtering accordingly.

I ended up writing something like the following:

```
  My_addr = 0;

  wiringPiSetup();			//setup wiringPi library
  cc1100.begin(My_addr);			//setup cc1000 RF IC
  cc1100.set_ISM(0x01);
  cc1100.receive();
  cc1100.set_channel(0x00);
  cc1100.wor_disable();//No wor mode
  cc1100.set_sync_mode(0x00);//No sync word

  cc1100.set_myaddr(0x00);//No address checking
  cc1100.set_data_whitening(0x00);//No data whitening
  cc1100.set_fec(0x00);//No FEC
  cc1100.set_preamble_len(0x00);//0 preamble length

  cc1100.spi_write_register(PKTCTRL1, 0x00);//Disable addr check

  cc1100.show_register_settings();
  for (;;) {
    delay(1);
    uint8_t bytes_in_RXFIFO  = cc1100.spi_read_register(RXBYTES);

    if((bytes_in_RXFIFO & 0x7F)){
      std::cout << "\nBytes to read!";
    }

    std::cout << "\n" << +bytes_in_RXFIFO;
  }
```

Despite this, it always logs out 0's despite using an RF device near it (and seeing it show on something such as RTL-SDR).

Any idea how I can accomplish this?

Thank you in advance.

gkchim, 2021-11-18:
Were you able to achieve being able to replay certain signal without knowing anything apart from the fact that it is 868mhz. if so how did you do it. I am trying to learn to do the same thing

adminy, 2024-08-01:
I think you're supposed to set receive registers after you set up all the options, not before. Everything else looks fine to me.