SpaceTeddy/CC1101 #52: Problems with Raspbian Bullseye

ulde1, 2022-01-14:
I try to compile your examples under Raspbian Bullseye, but there are two Issues.

## 1. wiringPi is not available any more.
 
My solution:
~~~~
git clone https://github.com/WiringPi/WiringPi
cd WiringPi
./build
~~~~
That seems to do the job.

## 2. Compile CC1101-Example results in linking errors
~~~~
git clone https://github.com/SpaceTeddy/CC1101
cd CC1101
cp ~/CC1101/examples/raspi/*.cpp ~/CC1101
sudo g++ -lwiringPi RX_Demo.cpp cc1100_raspi.cpp -o RX_Demo
~~~~
This will lead to linking errors:
~~~~
/usr/bin/ld: /tmp/ccFh2Z7u.o: in function `main':
RX_Demo.cpp:(.text+0x4e8): undefined reference to `wiringPiSetup'
/usr/bin/ld: RX_Demo.cpp:(.text+0x528): undefined reference to `delay'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::reset()':
cc1100_raspi.cpp:(.text+0x18): undefined reference to `digitalWrite'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0x20): undefined reference to `delayMicroseconds'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0x2c): undefined reference to `digitalWrite'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0x34): undefined reference to `delayMicroseconds'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0x48): undefined reference to `delay'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::wakeup()':
cc1100_raspi.cpp:(.text+0xa0): undefined reference to `digitalWrite'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0xa8): undefined reference to `delayMicroseconds'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0xb4): undefined reference to `digitalWrite'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0xbc): undefined reference to `delayMicroseconds'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::begin(unsigned char volatile&)':
cc1100_raspi.cpp:(.text+0x154): undefined reference to `pinMode'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0x160): undefined reference to `pinMode'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0x1b8): undefined reference to `delayMicroseconds'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0x1cc): undefined reference to `delayMicroseconds'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::sidle()':
cc1100_raspi.cpp:(.text+0x55c): undefined reference to `delayMicroseconds'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::transmit()':
cc1100_raspi.cpp:(.text+0x5c8): undefined reference to `delayMicroseconds'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::receive()':
cc1100_raspi.cpp:(.text+0x634): undefined reference to `delayMicroseconds'
/usr/bin/ld: /tmp/ccK6h5Rx.o:cc1100_raspi.cpp:(.text+0x6d8): more undefined references to `delayMicroseconds' follow
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::sent_packet(unsigned char, unsigned char, unsigned char*, unsigned char, unsigned char)':
cc1100_raspi.cpp:(.text+0xa80): undefined reference to `delay'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::packet_available()':
cc1100_raspi.cpp:(.text+0xba0): undefined reference to `digitalRead'
/usr/bin/ld: cc1100_raspi.cpp:(.text+0xbec): undefined reference to `digitalRead'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::wait_for_packet(unsigned short)':
cc1100_raspi.cpp:(.text+0x1128): undefined reference to `delay'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::spi_begin()':
cc1100_raspi.cpp:(.text+0x1a78): undefined reference to `wiringPiSPISetup'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::spi_write_register(unsigned char, unsigned char)':
cc1100_raspi.cpp:(.text+0x1b14): undefined reference to `wiringPiSPIDataRW'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::spi_read_register(unsigned char)':
cc1100_raspi.cpp:(.text+0x1b70): undefined reference to `wiringPiSPIDataRW'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::spi_write_strobe(unsigned char)':
cc1100_raspi.cpp:(.text+0x1bc4): undefined reference to `wiringPiSPIDataRW'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::spi_read_burst(unsigned char, unsigned char*, unsigned char)':
cc1100_raspi.cpp:(.text+0x1cac): undefined reference to `wiringPiSPIDataRW'
/usr/bin/ld: /tmp/ccK6h5Rx.o: in function `CC1100::spi_write_burst(unsigned char, unsigned char*, unsigned char)':
cc1100_raspi.cpp:(.text+0x1e24): undefined reference to `wiringPiSPIDataRW'
collect2: error: ld returned 1 exit status
~~~~

Can you pleas help me?

SpaceTeddy, 2022-03-30:
I do not support the lib anymore because of missing HW and SW setup, sry.

may you try to us gcc instead of g++ please. Hopefully it works.

kr,
Chris

no1knows, 2022-05-26:
Any luck here?

luciocorrea, 2022-08-20:
@no1knows try this:

$ g++ RX_Demo.cpp cc1100_raspi.cpp -o RX_Demo -lwiringPi

JakubLitwinowicz, 2022-10-23:
> @no1knows try this:
> 
> $ g++ RX_Demo.cpp cc1100_raspi.cpp -o RX_Demo -lwiringPi

U ARE A KING MAN, thank you. 1 hour through this bs and only your solution worked.