SpaceTeddy/CC1101 #56: No output after 'Init CC1100...' (Arduino)

DrahcPro, 2023-09-21:
I have installed the package and library, followed the instructions to install the script onto an arduino but it only prints "Init CC1100..." and nothing else. I've already jumped into the code at it must be going wrong somewhere between line 374 and 405 because after that there should be an output. Any tips on how to debug this further?

I am using an arduino mega 2560 together with an E07-M1101D-SMA chip.

closed 2023-10-03 by DrahcPro

SpaceTeddy, 2023-10-03:
Hi @DrahcPro ,
Does it mean you fixed your problem and it is working now? 
may you tell us the issue?

thx

DrahcPro, 2023-10-03:
The specific issue is not the case anymore (probs user error), I still have other issues which I am trying to figure out though. Namely tx side example is working but rx side is not receiving

DrahcPro, 2023-10-03:
Alright, I'll check that first 👍

DrahcPro, 2023-10-03:
Ow and now I have your attention, could you maybe add a note in the readme that the pinout for arduino MEGA is not the same as UNO this caused a fair bit of confusion for me as well (SPI are on the UNO in the 10s and in the MEGA on the 50s)